jamincollins/k2-improvements #2: OPKG update

kyfilm, 2025-01-08:
`root@K2Plus:/usr/share/mainsail# opkg update
Downloading http://bin.entware.net/armv7sf-k3.2/Packages.gz
wget: applet not found
*** Failed to download the package list from http://bin.entware.net/armv7sf-k3.2/Packages.gz

Collected errors:
 * opkg_download: Failed to download http://bin.entware.net/armv7sf-k3.2/Packages.gz, wget returned 127.`

jamincollins, 2025-01-08:
Can you provide the output of the following commands:

```
echo $PATH
find /opt/libexec
ls -l /opt/bin
```

onemorething3d, 2025-01-09:
```root@K2Plus:/mnt/exUDISK/k2-improvements-main/entware# opkg update
Downloading http://bin.entware.net/armv7sf-k3.2/Packages.gz
wget: applet not found
*** Failed to download the package list from http://bin.entware.net/armv7sf-k3.2/Packages.gz

Collected errors:
 * opkg_download: Failed to download http://bin.entware.net/armv7sf-k3.2/Packages.gz, wget returned 127.
root@K2Plus:/mnt/exUDISK/k2-improvements-main/entware# echo $PATH
/opt/bin:/opt/sbin:/usr/sbin:/usr/bin:/sbin:/bin
root@K2Plus:/mnt/exUDISK/k2-improvements-main/entware# find /opt/libexec
find: /opt/libexec: No such file or directory
root@K2Plus:/mnt/exUDISK/k2-improvements-main/entware# ls -l /opt/bin
total 740
-rwxr-xr-x    1 root     root        752572 Jan  9 09:48 opkg
root@K2Plus:/mnt/exUDISK/k2-improvements-main/entware#

jamincollins, 2025-01-09:
Looks like the install really didn't do anything but grab the opkg binary. Try rerunning the `generic.sh` script directly.

onemorething3d, 2025-01-10:
> Looks like the install really didn't do anything but grab the opkg binary. Try rerunning the `generic.sh` script directly.

I did a few times, but the results are the same.
'wget' error occurs.
It's the same as what I sent you on Discord last time.

jamincollins, 2025-01-10:
Please provide both the full `generic.sh` script that you're running and the full output of an attempt to run it.

onemorething3d, 2025-01-16:
I replaced the motherboard due to another error, reinstalled it and it works

closed 2025-01-16 by onemorething3d