jamincollins/k2-improvements #5: Bed mesh now probes back-right corner many times at the end

smartpbx, 2025-01-25:
Now when i complete a full bech mesh either from the fluidd interface, or by checking Calibration from CP, the full bed mesh runs, but at the end it probes the back right corner 5-10 times and makes a huge peak on the bed mesh

[image]

jamincollins, 2025-01-26:
This is due to the prtouch's variance. In our testing it generated wildly odd price results somewhere on the order of about 0.01% of the time. Sadly there is nothing we can do to compensate for the probe.

jamincollins, 2025-01-26:
This is a 37x37 mesh. You can see a number of such anomalous points on the mesh. 

[image]

smartpbx, 2025-01-26:
Thanks,

I only seem to get this point now after installing the improvement scripts and it does the strange probe in that spot. I have run the bed mesh 5 times with the standard points, and i have no anomolies like that, except the probe stopping at the last point and hitting it 5-10 times in a row. Ill try to catch a video of it.

dwaite, 2025-01-31:
I would also check the magnet on that corner - it may be peeling up from the plate.

closed 2025-02-19 by jamincollins