jamincollins/k2-improvements #6: start_print completely broken after updating

joshs85, 2025-01-26:
How am I supposed to use this?  I installed it and now when I print from fluid (a file that was sliced in orca) the nozzle just crashes into the walls at the start and then the print fails.  Do I still use the gcode from Creality print which is below?

M140 S0
M104 S0 
START_PRINT EXTRUDER_TEMP=[nozzle_temperature_initial_layer] BED_TEMP=[bed_temperature_initial_layer_single] CHAMBER_TEMP=[overall_chamber_temperature]
T[initial_no_support_extruder]
M104 S[nozzle_temperature_initial_layer]
M204 S2000
G1 Z3 F600
M83
G1 Y150 F12000
G1 X0 F12000
G1 Z0.2 F600
G1 X0 Y150 F6000
G1 X0 Y0 E15 F6000
G1 X150 Y0 E15 F6000
G92 E0
G1 Z1 F600

jamincollins, 2025-01-26:
Can you upload/attach the sliced gcode?

joshs85, 2025-01-26:
That was weird.. I re sliced it and restarted the machine and it seems to work fine..  I have a weird one.. maybe you can add something to fix.. When the printer powers on and checks the z tilt with the optical sensors.. my right side is always low by like 2-4 mm which causes my bed to be really out of wack. When I did screw tilt, I had to adjust the right side almost all of the way up.  Is there a way to add a static offset somewhere?

joshs85, 2025-01-26:
Also, Also, if you haven't already, you should change forced leveling to false so the user can select if they want to do the calibration or not with every print.

[virtual_sdcard]
path: /mnt/UDISK/printer_data/gcodes
forced_leveling: false

closed 2025-02-01 by jamincollins