jamincollins/k2-improvements #9: Over Heating Chamber

joshs85, 2025-02-01:
When I print with PETG it overheats the chamber by 5 degrees and some times never gets back down to the desired temp so it sits there for hours doing nothing with the nozzle hot.

jamincollins, 2025-02-03:
The additional 5 degrees is intentional.  Can you provide screenshots, logs, or video of the issue you're seeing?  I'm not sure what it's causing it wait/hang.

joshs85, 2025-02-05:
I had changed out the carbon filters on my printer to HEPA filters with carbon pellets.. is it possible that that's trapping the heat in.. It seems like the print will hang for 20-30 mins after getting to temp.. The last couple prints I have done, would start as soon as I open the door for a few mins.

txt4nk, 2025-02-19:
Okay so I just ran across this exact same issue, with the exact same hepa filter/carbon mod. It DOES indeed overshoot the chamber temperature and insists on waiting to print until your initial chamber temp is achieved. Here's a workaround while Jamin works it out.

do a git pull so you get the latest M191.cfg
if you've not already, create a overrides.cfg (there's a script to do this for you)
copy the whole gcode macro "_create_mesh" into the overrides.cfg and removed the maximum temperature on the "temperature_wait". It should currently read "MAXIMUM={S+5}".

[image]

jamincollins, 2025-02-19:
@txt4nk you're confusing things a bit here as you're working from the _revamp_ branch and I believe this report is against the _main_ branch.

txt4nk, 2025-02-19:
ah apologies, assumptions are a b!tch lol..

May be time to upgrade to the revamp!

joshs85, 2025-05-16:
So is this resolved in the latest version?