jamincollins/k2-improvements #10: Installing Obico after Entware installed?

NeodymiumPhish, 2025-02-03:
Hey all, is there any guide or instruction set for getting Obico running? I've been slowly working my way through each error I come across, but I just realized there's an `install_creality.sh` script in the Obico repo. Its only options are Sonic Pad and K1/K1 Max, though. Just wanted to see whether this was possible from the K1 option or if there's any minor changes I can make to get it up and running.

jamincollins, 2025-02-03:
Why not simply use the option to install Obico from the menu?

[image]

NeodymiumPhish, 2025-02-03:
Welp... That's enough internet for me today! Apparently I'm just dumb and completely missed this menu... Thanks!

closed 2025-02-03 by NeodymiumPhish