jamincollins/k2-improvements #11: Classic Camera disabled?

NeodymiumPhish, 2025-02-08:
I see that the mjpeg streamer option exists but is disabled in the entware menu.sh. Is there something wrong with this or has it been hidden unnecessarily?

jamincollins, 2025-02-08:
Originally, I thought it could coexist with the existing camera feed. But further testing revealed that it conflicts. So, I've disabled it for now.

NeodymiumPhish, 2025-02-08:
Is it possible to use the mjpeg function/improvements instead of the other feature in the main script? I’m trying to get the camera to work from apps like Mobileraker/Obico instead.

jamincollins, 2025-02-08:
Yes you only need entware and the classic camera bits. Obico needs other fixes that can be found in my PR against Obico.

Be aware that doing this will break all Creality camera integrations.

NeodymiumPhish, 2025-02-09:
Ah, understood! And is there anything we need to do if we already installed the camera mod from the main script?

Thanks for your help, just trying to make sure I understand fully what's necessary!

jamincollins, 2025-02-11:
That would be do to the pull bot force pushing, ugh..

jamincollins, 2025-02-11:
Should be fixed now:
https://github.com/TheSpaghettiDetective/moonraker-obico/pull/115

closed 2025-02-19 by jamincollins