jamincollins/k2-improvements #12: [Unverified] M191 Macro with 40C chamber temp causes 105C bed temp

AWildBeard, 2025-02-10:
Using Creality Print 5.1.7, setting the filament settings for heated chamber to 40C, with a 45C bed, appears to cause the printer to set the bed temperature to 105C during the temperature setting phase on the print. Setting the bed to 105C is not in the gcode of the sliced file as far as I can tell. This happens after the printer attempts what I believe to be a extrusion test.

Below is a screenshot showing the temperature spike.

[image]

jamincollins, 2025-02-10:
Yes, using my `M191` macro _and_ setting any chamber temp value will cause the bed to be heated to 105c if it is not already set to 100c or more:
https://github.com/jamincollins/k2-improvements/blob/main/features/chamber/m191.cfg#L12-L28

This is due to the anemic chamber heater on the K2.  However, my `START_PRINT` macro _will_ wait for the bed to return to the original desired temperature:
https://github.com/jamincollins/k2-improvements/blob/main/features/start_print/start_print.cfg#L72-L81

Out of curiosity what filament are you printing?

AWildBeard, 2025-02-10:
I'm printing ColorFabb LW-PLA-HT, an active foaming PLA with a reputation for bad stringing. 
I was wanting to use a 40-48C chamber temperature because supposedly that'll help with the stringing. Not sure I believe that snake oil. A 45C bed has been perfect to prevent warping on first layer or bed separation during print for me.
Getting the chamber to 40C hasn't been a huge issue or problem for me with the stock heater. I've not done any chamber heating shenanigans outside of this though.  
-------- Original Message --------On 2/10/25 8:32 AM, Jamin Collins  wrote:
Yes, using my M191 macro and setting any chamber temp value will cause the bed to be heated to 105c if it is not already set to 100c or more:
https://github.com/jamincollins/k2-improvements/blob/main/features/chamber/m191.cfg#L12-L28
This is due to the anemic chamber heater on the K2.  However, my START_PRINT macro will wait for the bed to return to the original desired temperature:
https://github.com/jamincollins/k2-improvements/blob/main/features/start_print/start_print.cfg#L72-L81
Out of curiosity what filament are you printing?

[
{
"target": "https://github.com/jamincollins/k2-improvements/issues/12#issuecomment-2648175379",
"url": "https://github.com/jamincollins/k2-improvements/issues/12#issuecomment-2648175379",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"name": "GitHub",
"url": "https://github.com"
}
}
]

AWildBeard, 2025-02-10:
I just let the print go all the way through and it indeed did print on the 105C bed. 

It heated up to 105C like you said, cooled back down all the way to 45C then heated back up to 105C, did a extrude test into the waste chute, then started printing.

[image]

[image]

closed 2025-02-19 by jamincollins