jamincollins/k2-improvements #13: printer fails to load existing mesh when sending print from CP6

Cboxhero, 2025-02-18:
I'll preface this with the fact that I do NOT know if this happens if the print is initiated in any other way.
But more times than not, when I send a print to the k2, it will run through the start_print until it hits: 
`RESPOND MSG="Loading bed mesh: {BED_TEMP}c_{CHAMBER_TEMP}c ..."
  BED_MESH_PROFILE LOAD={BED_TEMP}c_{CHAMBER_TEMP}c
  M117 MESH: {BED_TEMP}c_{CHAMBER_TEMP}c`

it will fail to load an existing mesh when one definitely does exist, proceed to create a new mesh at the default temp set in bed_mesh.cfg then home and idle. 
It will get stuck with a M117 instruction either saying it is heating the chamber or loading the mesh.
I'm not sure if this is some sort of conflict with CR6's filament setting "active temperature control" because when chamber is set to 0 and this is unchecked I don't run into this issue.

I've also noticed that when sending a print DOES work the nozzle is wiped 3 separate times and the z_comp is checked twice. Not entirely sure if this is intended, but ends up taking up quite a bit of time.

jamincollins, 2025-02-19:
Can you check the value of:

```
[virtual_sdcard]
forced_leveling: ...
```

Cboxhero, 2025-02-20:
> Can you check the value of:
> 
> ```
> [virtual_sdcard]
> forced_leveling: ...
> ```

[virtual_sdcard]
path: /mnt/UDISK/printer_data/gcodes
forced_leveling: true

jamincollins, 2025-02-20:
That is your issue. The true there is forcing a leveling.

closed 2025-03-24 by jamincollins