jamincollins/k2-improvements #17: Bed temp not recognized during chamber heating

rafpigna, 2025-02-23:
During chamber heating (M191) my bed temp is set at 120 (and already reached and stable) and chamber at 50, but the script says that the bed temp is not high enough and set it to 105 so a lot of time waste to go to 105, then after the chamber temp is reached it goes agai to the correct 110 and heat soak again, 2nd time! (so 5 minutes the first time BEFORE the chamber heating, and second time after it).

jamincollins, 2025-02-23:
can you share a sliced gcode that replicates this?

rafpigna, 2025-02-23:
> can you share a sliced gcode that replicates this?

here it is
I reprinted it and did same thing

[K2+Plus+upper+window+wedge.stl_PC_34m7s.zip](https://github.com/user-attachments/files/18931336/K2%2BPlus%2Bupper%2Bwindow%2Bwedge.stl_PC_34m7s.zip)

rafpigna, 2025-02-24:
@jamincollins 
I think I found the issue

My gcode inital script has M104 S0 and M140 S0 as first things, like Creality default config and only then, the START PRINT string.

The actual gcode file is not really loaded at printing start, but only when the first time bed reaches it's temp.
At this point the M104 S0 and M140 S0 are loaded so the heaters shut off to reach 0.
Just after that, the M191 is loaded from the START_PRINT script , but sees that the the bed target temp is 0, so set it to 105, also if the bed is still at 108, because since it's turned off it's slowly become cold.

Fix is to delete the M104 S0 and M140 S0 from the start gcode, and just leave as first row the START PRINT..... string.

Morticon404, 2025-03-13:
Can confirm, above is the fix. I was experiencing the exact same issue. After deleting M104 S0 and M140 S0 from the machine start G-code it now keeps the nozzle and bed at proper temp while awaiting chamber heat. Thanks @rafpigna. @jamincollins Maybe add to the FAQ or readme?

jamincollins, 2025-03-13:
@Morticon404 @rafpigna are either of you using this filament setting?

[image]

Morticon404, 2025-03-13:
@jamincollins Yes

[image]

Morticon404, 2025-03-14:
Not sure if this is intended, but if you have any temp set for the chamber in the filament settings it will carry over to the start print macro regardless of ticking the box or not.  Definitely need to double check before starting a print.

jamincollins, 2025-03-24:
The intended way to handle the filament chamber temperature is by passing it as a parameter to `START_PRINT` for better coordination of heating.

closed 2025-03-24 by jamincollins