jamincollins/k2-improvements #26: Print Calibation doesnt work anymore

rafpigna, 2025-02-23:
Sending print with "print calibration" enabled, both via Creality Print or via the Printer Screen, cause this issue:

The printer stay still, it not raise the bed or do the homing, but starts to heat the bed.
When the bed temp is reached, it starts to heat the nozzle. 
When the nozzle temp is reached it starts to move and raises the bed and do all the homing stuff, then starts the mesh calibration.
At the end of the mesh, all stops. 
Nozzle and bed remains hot and the print just hangs there forever. 
Stayed for 30 minutes to see is something happens: nothing. 
Before the script install all was working.
I cant revert to stock now because I have some important prints to do, but I will try also this.

Fluidd console not showing errors or messages.
When the printer starts to heat the bed or nozzle, Fluidd console showed nothing and the printer was read as "idle".
Only when the printer started homing, fluidd started to show messages from the console.

EP-Sys, 2025-02-24:
I got the same thing

jamincollins, 2025-02-24:
I've updated the documentation to indicate that the auto calibration option is not supported.