jamincollins/k2-improvements #27: Request: Instructions on how to install other features

KireanR, 2025-02-25:
Hello,

Sorry if this seems a little out there but was wondering if you could provide a write up on how to install the other features like Axis twist and things of that nature. Sorry if it's already provided but all I'm seeing is a bit on the main page telling us there is a install sh file for each and not much else like does everything in that folder already need to be on the printer and things of that nature

jamincollins, 2025-02-25:
I'll see if I can work some text up for the main page, but for each feature you just need to call the `install.sh` for that feature.

So, for the axis twist it would be:

```
sh ~/k2-improvements/features/axis_twist_compensation/install.sh 
```

KireanR, 2025-02-25:
Well got it installed and got a key342 error, option speed is not valid

emersonsc, 2025-02-28:
> I'll see if I can work some text up for the main page, but for each feature you just need to call the `install.sh` for that feature.
> 
> So, for the axis twist it would be:
> 
> ```
> sh ~/k2-improvements/features/axis_twist_compensation/install.sh 
> ```

Any time i tried to do that I got "file doesn't exist" even though I could confirm it was there with the ls command.

jamincollins, 2025-03-01:
@emersonsc any chance you can give me a screenshot or copy/paste of the full terminal output?

KireanR, 2025-03-01:
@jamincollins Would you be able to assist with the "option speed not valid" error? I'm running the non carto version