jamincollins/k2-improvements #29: Factory reset improvements

shirgall, 2025-02-28:
While the instruction to ssh into the printer and run 

   echo "all" | /usr/bin/nc -U /var/run/wipe.sock

to factory reset the printer is great, I am wondering if there should also be a way to factory reset the CFS. Frankly doing a factory reset after a firmware upgrade is important on the k2 plus itself and should be a well-known practice.

I just did this because my printer started acting weird and I notice all the CFS filament settings were preserved, so it made me wonder. At some point we will be uploading filament profiles to the CFS and at some point we will mess them up. :)

emersonsc, 2025-02-28:
> While the instruction to ssh into the printer and run
>
> echo "all" | /usr/bin/nc -U /var/run/wipe.sock
>
> to factory reset the printer is great, I am wondering if there should also be a way to factory [...]

I had to actually do a full factory reset/wipe and remove everything this script did because my CFS units stopped working. My k2 acted like they weren't connected. As soon as I did the above mentioned wipe along with factory reset via Creality wiki and reflashing the stock firmware i got functionality back from my CFS.

jamincollins, 2025-02-28:
The CFS state is stored in an area that is not erased but the normal wipe. This is part of the reason I added an improved reset: 

https://github.com/jamincollins/k2-improvements/blob/main/scripts/factory-reset-improved.sh

shirgall, 2025-02-28:
Heh, so you anticipated my issue by a week! :)

closed 2025-02-28 by shirgall