jamincollins/k2-improvements #30: FEATURE REQUEST:  Adaptive Bed Mesh - don't put purge line at the edge of the printbed

HAndreasH, 2025-03-01:
Hi Jamin,
I don't know if this is a simple fix or even possible, but would it not be better to set the purge line at the edge of the adaptive mesh instead of the edge of the print bed.
regards
Andreas (AnHeTech)

jamincollins, 2025-03-02:
Right now the purge line is outside the `START_PRINT` in the slicer's machine start gcode.  A macro could be made that places the purge line/blob/etc closer to the print area, but doing so would require users update their silcer's machine start gcode and remove multiple lines.

HAndreasH, 2025-03-02:
Thank you very much for the answer and the clarification ;-)

txt4nk, 2025-03-16:
> Hi Jamin, I don't know if this is a simple fix or even possible, but would it not be better to set the purge line at the edge of the adaptive mesh instead of the edge of the print bed. regards Andreas (AnHeTech)

Andreas, I've submitted a pull request adding this functionality.

HAndreasH, 2025-03-17:
I have currently solved this by removing the purge line in the printer profile and using a skirt
But of course it would be interesting to have the function automated.

closed 2025-04-07 by jamincollins