jamincollins/k2-improvements #41: Improvement : update Screws_Tilt_Adjust to include a center reference point

QuackenAudio, 2025-03-13:
I couldnt add a pull request because I am "not a colaborator"

[image]

Just asking to add a center reference to the screws tilt adjust macro instead of making one of the movable screws the non-moving reference

[screws_tilt_adjust]
screw1: 175, 175
screw1_name: Home
screw2: 45, 28
screw2_name: Front Left
screw3: 307, 28
screw3_name: Front Right
screw4: 307, 314
screw4_name: Rear Right
screw5: 45, 314
screw5_name: Rear Left
speed: 150
horizontal_move_z: 10
screw_thread: CW-M4

mpaw, 2025-03-22:
@QuackenAudio You can fork the repo, make your change in the fork, and then create a PR against this repo

QuackenAudio, 2025-03-22:
I actually did end up doing that... just had to learn how to do that
this can be closed

closed 2025-03-22 by QuackenAudio