jamincollins/k2-improvements #45: Feature request: stop motors after finished + timeout

nagyrobi, 2025-03-14:
After longer prints, when they finish, the machine parks the head and lowers the bed a bit, but all motors remain engaged and most fans stay spinning infinitely. Agreed this is important to properly cool down after finishing the task.

However there are valid cases when you start the print and leave the machine unattended to finish, and your next chance to get back to the unit is after maybe half a day, it's a pity to have all this machinery tensioned for so much time.

Suggestion is to introduce a stop motors command after finishing any print task after a longer delay, say, 30 minutes. 
Note to prevent the stop this if the user starts another print  earlier (if print in progress, don't stop).

wernight, 2025-03-30:
Possibly related, for me the UI keeps showing that it's printing. Also I cannot stop using the UI. Also visibly something is using 100% of the UI's CPU still. fluidd says it's done.

I'm using Orca and my end print is the default: `END_PRINT` (with an empty line after).

QuackenAudio, 2025-05-08:
@nagyrobi

just add M84 to the end of your End Print macro or to your End print area in the machine settings of your slicer

for a wait delay after completing the print, you can add G4 codes,
for example - 1 second = 1000 
G4 P60000 (one minute)
G4 P1800000 (30 Minutes)

its baked in as part of the print, so if you add it to your end print or to your machine code in teh slicer, it has to do all this before the print completes....

klipper is not really a software that allows it to interperet that you want it to auto cancel if you try to send a new print while it doing a G4... if you set the time for 30 minutes, you would have to power cycle the printer or use the emergency stop to end the print because cancelling a print in klipper has to finish its last command before stopping (because its has to send the next command as "stop")

thx1200, 2025-05-09:
You can potentially instead use the Idle Timeout feature of Klipper to do what you are looking for.  I haven't looked this up on any of my Creality printers to see how it's configured, but I have done this on other printers.  It's typically in the printer.cfg.  https://www.klipper3d.org/G-Codes.html#idle_timeout