jamincollins/k2-improvements #48: Regarding SCREWS_TILT_CALCULATE

YogevBokobza, 2025-03-16:
Hi,
Do I need to do SCREWS_TILT_CALCULATE with a bed temperature of 60c? or no heat?
It's telling me to turn "Rear Right" anticlockwise 00:16 but I can't turn it as it is all the way locked.. what to do?

Thanks!

Talaxin, 2025-03-17:
You shouldn't need to use heat, as its just setting a baseline within a set tolerance.

If you cant move it ACW., Move all the others CW, effectively "lowering" your rear right screw.

jamincollins, 2025-03-27:
I'm sure that several people face this behavior because the K2 beds are not well shaped. Due to the inherent curve in most K2 beds, most will never be able to get a perfect adjustment.

YogevBokobza, 2025-03-28:
@jamincollins Related to that I saw that in the modified START_PRINT there is this part:
```
  # better safe than sorry, re-homing is a small price to pay
  G28
  # ensure bed is level
  # some users have reported that the bed does not raise uniformly from the bottom
  Z_TILT_ADJUST
  # rehome Z after tilt adjust
  G28 Z
```

Why are you doing z tilt here? I noticed that in the original START_PRINT, this is commented out.

jamincollins, 2025-03-29:
The G28 will home X, Y, and Z.  When Z is home for the first time it will drop to the bottom, then raise to the top. In practice, we've found that it almost never raises flat. Mine for example has a 0.6mm tilt. The Z_TILT_ADJUST ensures the bed is level side to side.

nagyrobi, 2025-03-31:
A too to help with the screws: https://www.printables.com/model/1237142-bed-adjustment-knob-for-creality-k2-parametric

mpaw, 2025-04-05:
> I noticed on my printer that if I tighten all the screws and run SCREWS_TILT_CALCULATE it looks like the Rear Right has an issue where I should be able to tight it more but it's not possible

@YogevBokobza you should tighten all the screws and then back them all out a turn or two. That way your starting point gives you calibration room in both directions.

closed 2025-04-07 by jamincollins