jamincollins/k2-improvements #50: Additional tips for multi-filaments

wernight, 2025-03-22:
First the main issue I had, and I'm surprised it's not mentioned here, if there is that I had to add to each filament these else it wouldn't head to the right temperture:

```
; filament start gcode
M109 S[nozzle_temperature]
```

This is what Creality Slicer does by default as well.

Then, I wonder if we could put some tips for CFS. I didn't try much yet but it'd be nice to see what people suggest as I heard that Creality Slicer gives good results but Cura defaults had a lot of colour bleeding issues.

jamincollins, 2025-03-24:
This is not an issue with the improvements, but seems to be a slicer issue.  Closing

closed 2025-03-24 by jamincollins

wernight, 2025-03-28:
I agree. Thought it'd be useful to mention at least that one as the official way is missing it and many users use Orca.