jamincollins/k2-improvements #57: regarding auto-calibration

DeepflashX, 2025-04-03:
you are stating:
"As a heads up these improvements are not compatible with Creality's auto-calibration. In our experience we get better results through manual tuning."

Does this mean none of the steps Auto Leveling, Belt Tension, Cutter Calibration, Input Shaping will work?
If I am not going to use, Cartograph, how is the leveling done then?

nagyrobi, 2025-04-03:
Yeah

jamincollins, 2025-04-08:
We are currently testing the various calibration options to give better feedback in this area.

nagyrobi, 2025-04-09:
Also see this: https://github.com/Guilouz/Creality-Helper-Script-K2/issues/2 appears to be a simple fix

kronflux, 2025-05-21:
@jamincollins is there any update on this? while manual tuning may produce better results, it would be much more desirable to have this functionality in-tact and be able to optionally tune manually. I would imagine the auto calibration could be called or tweaked in order to function correctly. I'm still trying to wrap my head around the setup of K2's firmware but just going on how this is set up, I'd have to imagine it's possible

kronflux, 2025-05-26:
> Likely won't be any movement on my side for this. I don't use the auto calibrations and am busy working on other changes that will likely make them less likely to work going forward.

Thanks for the reply!

Would it at least be possible to update the README to elaborate on what exactly is not functional?
"not compatible with Creality's auto-calibration" isn't particularly descriptive and I believe @DeepflashX was asking specifically, what exactly is not functional?
"Does this mean none of the steps Auto Leveling, Belt Tension, Cutter Calibration, Input Shaping will work?"

I would expand on this by asking, is this in relation to:
-Intitial machine setup

-Running these calibrations manually from the screen:
[image]

-OR is it only in regards to the Print Calibration in Creality Print?
[image]

Some of us are sitting on our Cartographer modules due to not much information on this, and fearful that it will massively complicate our lives, despite the benefits, and unfortunately no one has done an installation video or anything along those lines as of yet :(

JamAndCheese, 2025-05-26:
> Also see this: https://github.com/Guilouz/Creality-Helper-Script-K2/issues/2 appears to be a simple fix

broken link

tsmith1823, 2025-06-14:
> Likely won't be any movement on my side for this. I don't use the auto calibrations and am busy working on other changes that will likely make them less likely to work going forward.

@jamincollins i think everyone understands that you are working on other changes that will likely make "them" less likely to work going forward. I have seen this question asked multiple times on reddit as well and it still is not being answered. this is what is preventing me from downloading the fixes in the first place. 

Is it ALL of the auto calibrations in the main printer menu or does this only make the pa and flow calibrations not work at time of printing. I really don't understand why this is such a hard question to answer for you, it is making it look like you have no idea what you are talking about and that doesn't really make me confident in your work.

txt4nk, 2025-06-14:
Printer calibrations in the main menu work perfectly fine. The calibrations that do not work are the PA and flow calibrations at the start of the print.

RyzenSavior316, 2025-06-14:
Just a side note... I am also sitting on a cartographer and have already printed the mount ect... I have been mostly waiting for more info to come out. It would be useful to know specifically what doesnt work... and or how easy is it to revert back.. is it a simple factory reset and remove hardware? In all fairness though I haven't done my due diligence yet as I know there is an active thread on creality forums and have not yet inquired there... but it might be nice to have a faq or readme hear that addresses those type of details. All though I thank you for your hard work and realize this is likely a labor of love over your actual job. Maybe you could delegate it?

txt4nk, 2025-06-14:
The PA and flow calibrations in the start print do not work, the leveling, belt, cutter and input shaping work perfectly fine. 

Removing the improvements is as easy as a factory reset.

tsmith1823, 2025-06-14:
> Printer calibrations in the main menu work perfectly fine. The calibrations that do not work are the PA and flow calibrations at the start of the print.

thank you.. Ive been looking for an answer to this question for three weeks now. this really should be in the faq. it is so vague on what works and what doesn't.

txt4nk, 2025-06-14:
I'll PR the faq and make it as clear as possible when I have some time. Please realize that the group of us that are working on this, including jamin, have day jobs, lives and families.. the time and effort put into this is simply "when we can". Hope that is understandable.

txt4nk, 2025-06-14:
This is the first im seeing it, else I would have answered it earlier than today.

kronflux, 2025-06-16:
It is worth noting that you absolutely MUST do manual calibrations for decent print quality, and the Z Offset will 100% need to be manually adjusted, if you use the variable_offset_PROBE, it will be far too close to the bed.
also worth noting that the Z Tilt will cause a lot of issues while you're first setting up the Cartographer. You'll likely need to reboot the machine a few times during the process. It's VERY finicky, and a very hacky solution. If/When Creality releases the source for the K2, better solutions will come along, but for now it's a very rough setup process.

txt4nk, 2025-06-16:
Install per thr instructions, run a print with pla and baby step the z offset in fluidd, then after the print completes and you are happy with the first layer click the save offsrt button, then save_config. Its really not that hard..

kronflux, 2025-06-16:
You're both right and wrong there, installing per the instructions is hit or miss. If your bed is particularly taco shaped, the Z_TILT step offsets the bed too far, and causes the Z homing to flat-out fail, until the steppers are disabled, and the bed actually falls flat to the bottom(usually during a reboot, but can do manually)
This is what happened to me, and getting things set up took a very long time until I understood what was going on.

My message was targeted to the user who said they're holding off on installing pending more info, keep in mind that not all users are comfortable with this type of thing.
This is an *extremely* hacky implementation, and every machine is different.
It may be "not that hard" for you, but for others that may not be the case.

fred28tft, 2025-06-30:
Hello,
I'm having problems with the mapper and my Creality K2 Plus. When printing, my printer suddenly stops with error 2560 Creality, {"code":"key560","msg":"Lost communication with MCU 'scanner'"}. Can you tell me why?
I did everything correctly, in order: resetting the printer, and all the steps jaminCollins instructed. The mapper levels the bed correctly and the printer prints well. But after several layers, the printer shuts down with an error code, as I noted above this message: Lost communication with MCU.
The link I used for the mapper and my Creality K2 Plus: https://github.com/jamincollins/k2-improvements/tree/main

Please tell me why.

Best regards,

fred28tft, 2025-06-30:
Does it work with the creality epoxy plate because I calibrated the mapper with the PEI plate and during my first print it worked because I put a PEI plate and during my last 2 prints which crash my printer error code 2560, {"code » : "key560 ","msg » : "Loss of communication with the MCU 'scanner'"} I was using an epoxy plate.