jamincollins/k2-improvements #58: issues after installation

DeepflashX, 2025-04-03:
I followed all the steps, including resetting via terminal and installed the improvements following the instructions.
Now when trying to print something, it bed levels forever and does not come to an end.
Any ideas what to do?

Jacob10383, 2025-04-04:
The bed leveling is taking forever because the improvements changed the default bed mesh from 9x9 to 19x19, which creates a much more accurate mesh at the expense of time. You can change this back to 9x9 via editing the overrides.cfg. However, if you save the printer cfg, and start new prints with the same bed temp, it will reuse this mesh so there is no bed leveling at all. In contrast, in my experience the stock firmware always re bed leveled upon reboot even if the mesh was saved to the cfg.

Issue 2 can be solved by homing the printer.

nagyrobi, 2025-04-04:
Recent firmware releases don't re-level the entire bed, just the area wehere the print lays.

DeepflashX, 2025-04-04:
@Jacob10383 thx for your reply. 
I think I tried that as well but the display simply was not reacting any more.
I did a fresh install now and only selected features I want/need, leaving out axis twist, entware, obico, secure auth, skip setup, m191, overrides, start_print and filaments_server (different project).
First 2 prints went good. Will add filament_server next.

Jacob10383, 2025-04-04:
> Recent firmware and creality print releases don't re-level the entire bed, just the area wehere the print lays.

Oh nice, that's good to know.

> [@Jacob10383](https://github.com/Jacob10383) thx for your reply. I think I tried that as well but the display simply was not reacting any more. I did a fresh install now and only selected features I want/need, leaving out axis twist, entware, obico, secure auth, skip setup, m191, overrides, start_print and filaments_server (different project). First 2 prints went good. Will add filament_server next. Is it possible to store more then 1 mesh for each possible bed temperature that then will be taken to avoid revelling for each temperature change?

Yeah, if u start a new print with a temp you don't have a mesh for, it will make another one and save it with that temp as the name. You can also use the "mesh_if_needed" macro, which takes in a temp and then creates a mesh for that temp if it does not exist. Just always remember you need to hit the save cfg&restart button at some point otherwise the mesh will be lost on a reboot. The screen not working is a weird issue. I had that for a while(but homing fixed it), but now my screen always works without homing. I've done a lot of tinkering so I might of accidentally fixed it at some point, but have no clue.

mpaw, 2025-04-04:
@Jacob10383 The screen not working seems to be related to the increased mesh size. I had the same issue with the screen not working until I leveled. I then commented out the mesh size in the overrides file
```
#[bed_mesh]
#probe_count:19,19
```
And since then haven't had any issues with the screen not working. Seems weird that the screen UI would be related to the mesh size, but considering how messy the Creality code is, I'm not surprised.

Jacob10383, 2025-04-04:
> [@Jacob10383](https://github.com/Jacob10383) The screen not working seems to be related to the increased mesh size. I had the same issue with the screen not working until I leveled. I then commented [...]

Hmm, that's an interesting observation. I did not comment it out but changed it to 11,11, so maybe it is just certain bigger meshes..?

genomez, 2025-04-05:
I changed mine to 11,11, deleted the previous saved meshes (this part was required), used mesh_if_needed, save config and reboot. Now, the screen is responsive on reboot and will allow me to home from there instead of requiring me to do it via fluidd. Thanks!

koenfaro90, 2025-04-07:
I'm pretty sure large mesh sizes screw up the printers initialization sequence making the printer only responsive after a manual home from fluidd. Probably they wait a certain amount of fixed time to load the mesh and loading a larger mesh takes longer.

Encountered this while increasing the mesh size without using this repo, and again after reset and installing this repo.

closed 2025-04-07 by jamincollins