jamincollins/k2-improvements #61: CFS detection failure

black8592, 2025-04-08:
First install was unclean, CFS not detected right away.
Second install I did a factory reset before, worked one print, gave me an error aber filament beeing low and CFS was gone again.

I used the no carte installer with K2 and CFS up2date.

black8592, 2025-04-10:
V1.1.2.10 

i had the default profile and one for abs, made no changes on files

chkathome, 2025-04-10:
Try to reduce the probe count in "custom/overrides.cfg" to "probe_count:9,9" and delete all saved meshes in "printer.cfg".
Restart and do a mesh with the new setting.
This did the trick for me and I had at least 4 meshes saved without loosing the connection to the CFS.
Now I use the Carto and don't need to save meshes anymore...

closed 2025-05-05 by jamincollins