jamincollins/k2-improvements #69: freezes the on board touch screen

sxenth, 2025-04-28:
used the install script for no carto and installed with no errors. the touch screen works fine when the printer turns on but if you set up a print, at some point it freezes in the sense that it does not display current temps and you can not interact with the printer. you can still move around in the touch screen ui but if you want to stop the print for example once you hit stop nothing happens. the issue seems to be somewhere around when it sets up for a heat soak.

sxenth, 2025-05-01:
have been testing with this and as far as I can tell the problem is coming from the bed mesh causing some sort of communication issue with the touch screen. I've been testing with 2 k2 plus machines one set up with pla and the other set up with abs and a chamber temp of 60 both using the same printer profile in orca and the same stl model just different filament profiles. 

the touch screen seems to be fine and accurate all the way through till the end of the bed mesh and then ether freezes only showing the information that it had when it made its last probe from the bed mesh or defaults back to the temp graph and shows 0 for all the temps but still allowing you to move around the ui in both cases. the only fix I have found for the first case is to power down and restart and for the second you can home the machine and its fine and this is consistent with both machines.

 if for what ever reason it does not have to do a bed mesh (ether the profile exists or oddly enough if you loose power and do a power restore resume print) then the touch screen works fine, also consistent for both machines.

I have had some success by adding m118 and dummy moves to the end of the start print, create mesh and mesh if needed macros. or by lowering the probe count for the bed mesh although for some reason what works for one machine does not work for the other even though they are both on the same firmware with the same cfg files, the only thing that I have found that works on both machines is lowering the probe count to 9,9

jamincollins, 2025-05-01:
Are you using the stock probe (prtouch)?

sxenth, 2025-05-02:
Yea other than the no carto install script both machines are completely stock. I did read in the faq about the temps dropping on the touch screen and fluidd occasionally hanging at 99% as well and found that lowering the probe count to 9,9 with the no carto install also prevented both those issues as well as the touch screen issue I was experiencing. I do have 2 cartographers ordered and on the way but wanted the benefit of the machine waiting for chamber temps before bed meshing while I waited.

jamincollins, 2025-05-02:
it's a hit or miss issue for those with the prtouch.  Some (like myself) were able to go well above the original 9x9 mesh, some are restricted to 13x13 (IIRC) or less.

closed 2025-05-02 by jamincollins