jamincollins/k2-improvements #71: UI stuck after printing

wernight, 2025-05-05:
Steps:

1. Installed on stock K2 Plus without cartographer, on top of firmware 1.1.2.10; with CFS.
2. Print from Orca (basically same as using fluidd)

Expected:

 - After printing, allows touch UI.

Actual:

 - Fluidd works and just shows "M117 Printing ..." under the "Ready" tab.
 - Stuck in "M117 Printing ..." with both touch UI and mobile UI showing it's printing.
 - Mobile UI says "Nozzle TEMP 99%".

Tried:

 - Printing with and without enabling CFS (gives similar results)
 - fluidd > Home Z (works but doesn't unblock)
 - fluidd > Home XY (works but doesn't unblock)
 - fluidd > Services > Restart Klipper (fails: Can not update MCU nozzle_mcu config as it is shutdown)
 - fluidd > Host > Reboot (works)

jamincollins, 2025-05-05:
Are you able to send another print?

wernight, 2025-05-05:
Yes via fluidd. Fluidd basically looks like the print is complete I think.

DeepflashX, 2025-05-07:
don't install overrides.cfg and bed_mesh. 
so go back to factory default, and only install the rest. For some reason big meshes fuck something up.

wernight, 2025-05-07:
I see `overrides.cfg` with mostly default settings for bed mesh.

So basically are saying to not use the `no-carto.sh`, as [it has bed mesh](https://github.com/jamincollins/k2-improvements/blob/d9fef11151959649084dbf546efd9db11f716efb/no-carto.sh#L22), but instead install everything except bed_mesh?

The main reasons I installed k2-improvements, is actually to have bed mesh be done on the warmed up bed. If I understand, you're suggesting to disable completely the [bed mesh macro](https://github.com/jamincollins/k2-improvements/blob/main/features/macros/bed_mesh/README.md), it removes pretty much all my reasons to install it. Not a blame, just stating my POV; so I'd rather help on fixing the issue.

DeepflashX, 2025-05-07:
I use it like that from jamincollins for many other positive aspects + this: https://github.com/MasterLufier/Creality-K2-Custom-Macros

Gives me 11x11 bed meshes for each bed temperature and reuses those. That works fine..

Stelk, 2025-05-07:
I'm going to +1 this issue.  I have very similar circumstances.  I used the no-carto.sh install option and my on device ui stopped responding.  One difference from the OP scenario is that I could no longer print from creality print.  It would always do a 9x9 bed mesh and then terminate the build after the probing was completed.  No error in any logs or on console that  I could find.  

Factory reset fixes the problem.. 

Thanks

JPFASTLANE, 2025-06-25:
I think i'm having the same issue. My issue is only at the printer Touch LCD stops working correctly after a prints has been started at a random time in to the print. The temps are not updating and you can not stop or pass the printer. It all works from Fluidd web UI fine.